sony/flutter-embedded-linux #290: Poor render performances on iMX 8 Nano (X11)

Tom3652, 2022-09-16:
Hello and thanks again for the package !

I am having trouble to see good performances on my linux device. 
I am running my flutter app (`release` mode) on an [IMX 8 Nano](https://www.nxp.com/products/processors-and-microcontrollers/arm-processors/i-mx-applications-processors/i-mx-8-processors/i-mx-8m-nano-family-arm-cortex-a53-cortex-m7:i.MX8MNANO) under Debian 11.

My device has no GPU, 1 CPU with 1 core -  1Go of RAM - 1Go disk space.
I run the flutter app under OpenBox in X11. 

Animation are very janky and missing a lot of frames (`Navigator.push` with the default animation), scrolling a ListView above other widgets is also laggy. 
The CPU is also rising up to 90% while scrolling a `ListView.builder`. 
Any `Transform.translate` animation style is janky. 

I would like to run a Wayland compositor but i can't find a good one : 

- Weston doesn't run standalone - DRM is not supported and under a Openbox - XFCE it works but the performances are worse.
- GNOME requires 2Go disk space that i have not.

I have seen that you have tested : 

[image]

May i know which Wayland compositor did you use please ? 
Do you think my device hardware specifications are enough (without GPU or only 1 core) to run a usual Flutter app ? 

Thanks a lot in advance

 Note : I have run a basic flutter app with only a `ListView.builder` inside, the scrolling is rising the CPU up to 90% but the animation is quite smooth on X11 (OpenBox). 
 If i scroll a `ListView.builder` above a `Container` that has a `Image.asset` behind (100x100 with 100Kb), it starts lagging.

HidenoriMatsubayashi, 2022-09-17:
First of all, if you want to improve CPU performance, you should not use X11 backend. Also, did you enable GPU hardware accelerator on i.MX8 Nano? Maybe, GPU performance of i.MX8 nano is not so good, I guess.

Tom3652, 2022-09-17:
Thank you very much for your answer, no i have not enabled it i will check how to do so ! 

Weston doesn't work on my IMX 8 unless i run inside another compositor such as XFCE (for a Wayland backend)

Is that the way you have used it ?

Tom3652, 2022-09-26:
Could you please answer me on this last question @HidenoriMatsubayashi ? 
Would be much appreciated

psstoyanov, 2022-09-26:
Try with [Cage](https://github.com/Hjdskes/cage) - it's minimal kios-style of compositor based on [wlroots](https://github.com/swaywm/wlroots).

What you are doing, is running Weston as "[nested](https://gitlab.freedesktop.org/wayland/weston/-/blob/master/README.md#running-weston)" session - you will get the overhead of the X11 session which you don't intend to use.
Ideally, you wouldn't nest different sessions - stick to just one type to lower the variables and used resources.

If I were you, I would try the following:
- Check if you can run the Flutter app directly in DRM mode -  check here: https://github.com/sony/flutter-embedded-linux/wiki/How-to-run-Flutter-apps#2-run-with-drm-backend This ought to be least resource intensive method.

- Try running standalone Wayland session - Weston or Cage (I find Cage easier to use and compile but your mileage can vary depending on the board)

- Use the X11 session directly instead of nesting both X11 and Wayland session

Tom3652, 2022-09-26:
Thank you @psstoyanov for taking time to answer me. 
I will try Cage and tell you how it goes ! 

> Check if you can run the Flutter app directly in DRM mode - check here: https://github.com/sony/flutter-embedded-linux/wiki/How-to-run-Flutter-apps#2-run-with-drm-backend This ought to be least resource intensive method.

Unfortunately DRM is not supported on IMX 8 (i have tried running Weston Standalone with DRM backend on an VM and it works perfectly but not on my IMX). 

> Use the X11 session directly instead of nesting both X11 and Wayland session

This is what i am currently doing using Openbox with a standalone X11 session which is working but has no good performances, that's why i want to make some Wayland compositor working. 

The worst case scenario : i will keep my X11 ...

HidenoriMatsubayashi, 2022-09-26:
BTW, i.MX 8M nano doesn't provide BSP including Wayland/Weston?

Tom3652, 2022-09-26:
Thanks for suggesting this, i am looking into BSP (to know what is it)

ysimonx, 2022-09-26:
> Thanks for suggesting this, i am looking into BSP (to know what is it)

@HidenoriMatsubayashi 

I am trying to help @Tom3652 .
we have a iMX8 nano version, without any GPU

BSP seems to be present here : https://www.nxp.com/design/software/embedded-software/i-mx-software/embedded-linux-for-i-mx-applications-processors:IMXLINUX

for wayland/weston : I found 

- https://www.nxp.com/docs/en/reference-manual/IMX_REFERENCE_MANUAL.pdf
(page 124)

- https://www.nxp.com/docs/en/user-guide/IMX_GRAPHICS_USERS_GUIDE.pdf
(page 78)

but, we don't understand where (or how to build) the graphic drivers can be found

can you help us please ?

Tom3652, 2022-10-14:
Thanks for the feedback.

Another member of my team told me that my flutter app needed to be compiled with an external module that allows communication with the GPU somehow. 
The binary file looks like this :  `imx-gpu-viv-6.4.3.p1.0-aarch64.tar.gz`

Is it something i really need to do ? 
If yes, how can i add this external module when building the flutter app using for example `flutter-elinux build elinux --target-backend-type=x11` please ?

psstoyanov, 2022-10-14:
That binary file seems to be a kernel module for the board.

Better to take a look at: https://community.nxp.com/t5/i-MX-Processors/GPU-source-code-with-opencl/m-p/1224451/highlight/true#M168784

Tom3652, 2022-10-14:
Thanks for your reply @psstoyanov indeed it's the GPU driver module as far as i understand. 

So does it mean it should not be compiled along with the Flutter build, but rather with a specific "backend" ? 
I see many times Yocto coming back in my researches and again in the link you provided, what role it has to play here ?

psstoyanov, 2022-10-14:
I'm not familiar with Yocto but as far as I understand it, you have a "recipe" to build an image for the board and what packages to include in it.

Taking a second a look in the link, the suggestion is to place the binary gpu lib in the expected location from the official `rootfs`.
At a glance, you would be looking for the driver for `Vivante GC7000UL GPU` but do check the board specs again.

I may be wrong depending on how wayland/ X11 is compiled from the recipe but adding the relevant lib should be pretty much it 

iirc as I'm trying to get Lakka LibreElec with Wayland instead of X11, the scripts are automatically trying to figure out which drivers are available for the board and make adjustments based on that.
I don't expect that you would have that problem if you use the OEM Yocto recipes or the provided images from [npx](https://www.nxp.com/).

Tom3652, 2022-10-14:
Thank you very much for your time, i will discuss this with my teammates and give a feedback !

Tom3652, 2022-10-14:
Does it have anything to do with [the custom artifacts build](https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps#how-to-build-libflutter_engineso-and-libflutter_elinux_so) ? 
I mean do i have to build the engine along with my driver using something like [mentioned here](https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps#how-to-build-libflutter_engineso-and-libflutter_elinux_so)
?
Or it will not change anything ?

psstoyanov, 2022-10-14:
I can't really say.
I'm not using similar setup i.MX board with proprietary GPU driver for now. I've never gotten around to setting up the Android Things i.MX7 board for Flutter development with elinux yet.
I don't expect that you would need an entirely custom Flutter artifact but I could be completely wrong here.

I would start by making sure the GPU kernel module is loaded and in use by the board first.

Tom3652, 2022-10-14:
Thank you for your confirmation, i understand your position and i agree, we will make sure of this first :)

HidenoriMatsubayashi, 2022-10-14:
glibc version will differ depending on target environments (BPSs), so I recommend you to build .so files using the toolchain for your environment after confirming the GPU kernel module.

See also: https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps#pre-built-images

Tom3652, 2023-03-14:
It is working very fine today using Weston and `wayland` backend !

closed 2023-03-14 by Tom3652