sony/flutter-embedded-linux #323: Flutter engine reload leads to incorrect rotation being applied

Taha-Firoz, 2023-01-24:
Firstly I'd like to wish you a Happy New Year @HidenoriMatsubayashi and thank you for all your hard work behind this project even after you've departed Sony. I've experienced this issue before but I would always figure out some work around but I unfortunately can't any more, it's become too frequent. 

I also have no idea why it's happening. I have a flutter application that is a client to some dbus services, we're using the `dbus.dart` crate for it. We've also applied a `270°` rotation to the application via the `-r 270` flag and everything works correctly. Previous rotation bug #256 was closed however we experience the exact same issue but after the application has started

1. The application starts in the correct landscape orientation with no issues
2. The application attaches it self to dbus services
3. If the dbus services are already up then everything remains normal
4. The application now rotates exactly this way
[image]

This rotation is technically correct and is different from the #256 bug, but it seems as if the height and width have been switched. I have gone through flutter-embedded-linux and I am unable to find an area where the DRM window is reconfigured and that too with different parameters. 

Initially we considered that this was anomalous and random, however after we took some steps to try to reproduce this we noticed it happens exactly in this order. 

For some reason sometimes if the order in which the dbus service comes up is a littler later we think the flutter engine redraws the ui but this time without the rotation parameters specified for some reason. 

I've checked logs and there aren't any exceptions in the logs, everything continues as normal for the flutter binary, it doesn't crash it doesn't get restarted except for just displaying the orientation incorrectly. I believe this may be an internal dart exception, something that's triggering the screen to re-render incorrectly. 

Since the #256 bug has already been closed do you know of any other instances where the draw area is possible reset or readjusted, I'm sure the problem is somewhat related to there

HidenoriMatsubayashi, 2023-01-24:
Hi, thank you for the issue report. But it looks like it's difficult for me to reproduce this issue. Can you reproduce this bug without using the dbus services?

Taha-Firoz, 2023-01-24:
I'll try to setup an MVP with a simple dbus service try to replicate the issue on it. Do you know how this could possibly be happening though and is there anyway we can gather more information for you?

HidenoriMatsubayashi, 2023-01-25:
Basically, no. I think I need to replicate the issue first locally and debug it. Can you please try to run your app using `FLUTTER_LOG_LEVELS=TRACE`?

e.g.
```
$ FLUTTER_LOG_LEVELS=TRACE ./flutter-client --bundle=<path_to_flutter_project_bundle>
```