sony/flutter-embedded-linux #326: Screen corrupts on Wayland

HidenoriMatsubayashi, 2023-01-25:
The screen is distorted with high rate. https://github.com/sony/flutter-embedded-linux/pull/314 seems to affect this bug.

[image]

HidenoriMatsubayashi, 2023-01-26:
It seems this symptom depends on Ubuntu 22 (GNOME).

swift-kim, 2023-01-31:
Is this issue possibly related to https://github.com/flutter/flutter/issues/119601? Does it only occur under some specific conditions (Ubuntu/Wayland)?

HidenoriMatsubayashi, 2023-01-31:
Could be. Thanks for the information. Actually, I was not able to reproduce this issue on X11 backend and Sway (Wayland compositor). I can reproduce it only on Ubuntu 22 (GNOME, Weston).

andreadaoud, 2023-02-18:
Can you patch this when building flutter engine in CI pipeline? From my testing, the following patch solves this issue:

```
diff --git a/shell/platform/embedder/embedder_surface_gl.cc b/shell/platform/embedder/embedder_surface_gl.cc
index 44b1fddf80..5b64ad2e28 100644
--- a/shell/platform/embedder/embedder_surface_gl.cc
+++ b/shell/platform/embedder/embedder_surface_gl.cc
@@ -88,8 +88,7 @@ SurfaceFrame::FramebufferInfo EmbedderSurfaceGL::GLContextFramebufferInfo()
   // Enable partial repaint by default on the embedders.
   auto info = SurfaceFrame::FramebufferInfo{};
   info.supports_readback = true;
-  info.supports_partial_repaint =
-      gl_dispatch_table_.gl_populate_existing_damage != nullptr;
+  info.supports_partial_repaint = false;
   return info;
 }
```

HidenoriMatsubayashi, 2023-02-21:
I'm considering adding populate_existing_damage support instead.

PR #331 (2023-02-25): add dirty region management
Fixed #326
commit c71ea47f (2023-02-25): add dirty region management (#331)

Fixed #326
closed 2023-02-25 by HidenoriMatsubayashi