sony/flutter-embedded-linux #329: "Add HighDPI support for Wayland" breaks fullscreen

andreadaoud, 2023-02-18:
https://github.com/sony/flutter-embedded-linux/commit/d88c3c55d9bce06a864f5b51a6d5a9a71293a336

Hello @vially ,

In my use case, my app is put into a kiosk weston compositor to be launched in fullscreen mode, and with scale factor is overrode (`./app --bundle=. --fullscreen --force-scale-factor=1.3`). My intention of the scale factor is only to scale UI elements. The screen size is 800x480. However, when this patch is applied, the screen is scaled to bigger than normal screen size. From the log, you can see that it is actually 1040x624, so some portion of the UI is displayed outside of the screen. When I revert this patch, all things works normally. Could you please propose a fix for my use case? I'm willing to test your solution.

```
# ./app --bundle=. --fullscreen --force-scale-factor=1.3
[TRACE][operator()(171)] presentation info: clk_id = 1
[INFO][operator()(606)] Display scale for output(8): 1
[INFO][operator()(573)] Display output info: width = 800, height = 480, refresh = 62413
[TRACE][CreateRenderSurface(1149)] Created the Wayland surface: 1040x624
[TRACE][operator()(143)] window entered output 8
```

vially, 2023-02-18:
Unfortunately I will not have access to a development machine for the next couple of weeks so I will not be able to investigate and debug this issue. This means I can only try to provide some context for what that commit did and what I think might be required to fix this.

> My intention of the scale factor is only to scale UI elements

I think this is the root of the issue. The changes from #314 assumed the `--force-scale-factor` flag to mean forcing the Wayland [_output_ scale factor](https://wayland.app/protocols/wayland#wl_output:event:scale) but I see now that it's meant to be an additional scaling factor that's independent of the Wayland's output/surface scale.

I guess the fix would involve changing all calculations on Wayland from the current form of:

```
xy_coordinate_in_px = xy_coordinate_in_dp * output_scale
```

to something like:

```
xy_coordinate_in_px = xy_coordinate_in_dp * output_scale * forced_scale
```

Of course, the hard part would be identifying all the places where this change is needed and plubming it through.

HidenoriMatsubayashi, 2023-08-11:
Currently, it looks like even mouse cursor (click) doesn't work correctly when using `--force-scale-factor`.