sony/flutter-embedded-linux #330: xdg_surface has never been configured

SinkinZhu, 2023-02-18:
Dear Sir

When we run our Flutter APP, it start failed, and show the error log : xdg_surface has never been configured
Can you help me introduce how to debug this issue, Thanks

HidenoriMatsubayashi, 2023-02-19:
What wayland compositor are you using?

Also, can you please share the whole log?
```
$ FLUTTER_LOG_LEVELS=TRACE ./flutter-client --bundle=<path_to_flutter_project_bundle>
```

makotosato-at, 2023-03-10:
I got the same issue.
This phenomenon also occurs in RPi4.

This phenomenon seems to occur under the following conditions:
1. Using weston.
2. Using video_player or/and camera plugin.
3. Application is built in release mode.
4. "_$HOME/.cache/gstreamer-1.0/_" directory does **not** exist.

If "_$HOME/.cache/gstreamer-1.0/_" exists, this phenomenon is not occur.
"_$HOME/.cache/gstreamer-1.0/_" is created by _gst_init()_.

I don't know the detailed reasons, but it works fine when I try the following:
<pre>
diff --git a/packages/camera/elinux/camera_elinux_plugin.cc b/packages/camera/elinux/camera_elinux_plugin.cc
index 4f841e0..c72f3dd 100644
--- a/packages/camera/elinux/camera_elinux_plugin.cc
+++ b/packages/camera/elinux/camera_elinux_plugin.cc
@@ -60,7 +60,6 @@ class CameraPlugin : public flutter::Plugin {
                flutter::TextureRegistrar* texture_registrar)
       : plugin_registrar_(plugin_registrar),
         texture_registrar_(texture_registrar) {
-    GstCamera::GstLibraryLoad();
   }
   virtual ~CameraPlugin() {
     if (camera_) {
diff --git a/packages/camera/elinux/gst_camera.cc b/packages/camera/elinux/gst_camera.cc
index 9bb0e47..0f383e8 100644
--- a/packages/camera/elinux/gst_camera.cc
+++ b/packages/camera/elinux/gst_camera.cc
@@ -8,6 +8,7 @@
 
 GstCamera::GstCamera(std::unique_ptr<CameraStreamHandler> handler)
     : stream_handler_(std::move(handler)) {
+  gst_init(NULL, NULL);
   gst_.pipeline = nullptr;
   gst_.camerabin = nullptr;
   gst_.video_convert = nullptr;
diff --git a/packages/video_player/elinux/gst_video_player.cc b/packages/video_player/elinux/gst_video_player.cc
index 02339e8..f4433f7 100644
--- a/packages/video_player/elinux/gst_video_player.cc
+++ b/packages/video_player/elinux/gst_video_player.cc
@@ -9,6 +9,7 @@
 GstVideoPlayer::GstVideoPlayer(
     const std::string& uri, std::unique_ptr<VideoPlayerStreamHandler> handler)
     : stream_handler_(std::move(handler)) {
+  gst_init(NULL, NULL); 
   gst_.pipeline = nullptr;
   gst_.playbin = nullptr;
   gst_.video_convert = nullptr;
diff --git a/packages/video_player/elinux/video_player_elinux_plugin.cc b/packages/video_player/elinux/video_player_elinux_plugin.cc
index a703e31..a41405a 100644
--- a/packages/video_player/elinux/video_player_elinux_plugin.cc
+++ b/packages/video_player/elinux/video_player_elinux_plugin.cc
@@ -58,9 +58,6 @@ class VideoPlayerPlugin : public flutter::Plugin {
                     flutter::TextureRegistrar* texture_registrar)
       : plugin_registrar_(plugin_registrar),
         texture_registrar_(texture_registrar) {
-    // Needs to call 'gst_init' that initializing the GStreamer library before
-    // using it.
-    GstVideoPlayer::GstLibraryLoad();
   }
   virtual ~VideoPlayerPlugin() {
     for (auto itr = players_.begin(); itr != players_.end(); itr++) {
</pre>

makotosato-at, 2023-03-23:
Hello.

I could not get `FLUTTER_LOG_LEVELS=TRACE` log because it only occurs in the release version, 
but I could get `WAYLAND_DEBUG=1` log instead.

flutter-elinux version is 3.3.10 and I got this log on RPi4.
[log.txt](https://github.com/sony/flutter-embedded-linux/files/11047006/log.txt)

HidenoriMatsubayashi, 2023-03-23:
Thanks.

Did mouse cursor is enabled when the error happened? If so, can you please try again using `--no-cursor` option as a test?

makotosato-at, 2023-03-24:
> Did mouse cursor is enabled when the error happened? If so, can you please try again using --no-cursor option as a test?

I tried it and it didn't work.
Again, I don't know why, but if I remove `$HOME/.cache/gstreamer-1.0`, it works fine.
It doesn't seem to have anything to do with weston...

HidenoriMatsubayashi, 2023-03-24:
> Again, I don't know why, but if $HOME/.cache/gstreamer-1.0 exists, it works fine.

Strange. Does the issue happen even if using only video or camera plugin only?

makotosato-at, 2023-03-27:
> Strange. Does the issue happen even if using only video or camera plugin only?

Yes.

On my RPi4, if `$HOME/.cache/gstreamer-1.0` is not exists, `gst_init()` takes about 1600ms to execute.
If `$HOME/.cache/gstreamer-1.0` exists, it takes 41ms.

This time may be causing the application initialization sequence to be incorrect.

As a test, `sleep(1)` instead of `gst_init()` also causes the phenomenon.
video_player_elinux_plugin.cc:
```
  VideoPlayerPlugin(flutter::PluginRegistrar* plugin_registrar,
                    flutter::TextureRegistrar* texture_registrar)
      : plugin_registrar_(plugin_registrar),
        texture_registrar_(texture_registrar) {
    // Needs to call 'gst_init' that initializing the GStreamer library before
    // using it.
    //GstVideoPlayer::GstLibraryLoad();
    sleep(1);
  }
```

HidenoriMatsubayashi, 2023-03-27:
I was not able to repro the issue. Can you please check if `xdg_surface_ack_configure()` is called before `wl_surface_commit` call?

See also: https://aznote.jakou.com/prog/wayland/06.html

makotosato-at, 2023-03-29:
> I was not able to repro the issue. Can you please check if xdg_surface_ack_configure() is called before wl_surface_commit call?

I attached a log (ng_log.txt) when the phenomenon occurs and a log (ok_log.txt) when it works correctly.
It seems that xdg_surface_ack_configure is called after wl_surface_commit.
ng_log.txt:
```
...
[4209127.458]  -> wl_surface@15.commit()
...
[4210790.425]  -> wl_surface@15.commit()
...
[4211134.920]  -> xdg_surface@17.ack_configure(205)
[4211135.022] wl_display@1.delete_id(24)
[4211135.061] wl_display@1.delete_id(26)
[4211135.094] wl_display@1.error(xdg_surface@17, 3, "xdg_surface has never been configured")
xdg_surface@17: error 3: xdg_surface has never been configured
```
[ng_log.txt](https://github.com/sony/flutter-embedded-linux/files/11098262/ng_log.txt)
[ok_log.txt](https://github.com/sony/flutter-embedded-linux/files/11098264/ok_log.txt)

HidenoriMatsubayashi, 2023-03-29:
@makotosato-at I'm not 100% sure, but can you please try https://github.com/sony/flutter-embedded-linux/pull/336 patch?

makotosato-at, 2023-03-30:
Thank you for the patch.
I tried but it didn't work.

HidenoriMatsubayashi, 2023-04-01:
I've update the patch. Can you please try with it again? Also, please try flullscreen (`-f`) option and without its option.

makotosato-at, 2023-04-04:
Thank you.
It works fine!!

> Also, please try flullscreen (-f) option and without its option.

Both ok.

PR #336 (2023-04-04): wayland: fix "xdg_surface has never been configured" error
Fixed #330
commit 393c4e09 (2023-04-04): wayland: fix "xdg_surface has never been configured" error (#336)

Fixed #330
closed 2023-04-04 by HidenoriMatsubayashi