sony/flutter-embedded-linux #332: eglSetDamageRegionKHR failed: eglGetError: EGL_BAD_ACCESS

makotosato-at, 2023-03-06:
Hello.

I am using an i.MX8MP board.
When I launch the application in Flutter 3.7.6, 
I get "eglSetDamageRegionKHR failed: eglGetError: EGL_BAD_ACCESS".
Is there any workaround for this?

The error log is below:
<pre>
[TRACE][operator()(171)] presentation info: clk_id = 1
[INFO][operator()(606)] Display scale for output(9): 1
[INFO][operator()(573)] Display output info: width = 1920, height = 1080, refresh = 60000
[TRACE][CreateRenderSurface(1148)] Created the Wayland surface: 1920x1080
flutter: The Dart VM service is listening on http://127.0.0.1:45875/eBjs9piUlmE=/
[ERROR][SwapBuffers(116)] eglSetDamageRegionKHR failed: eglGetError: EGL_BAD_ACCESS
[TRACE][operator()(143)] window entered output 9
[ERROR][SwapBuffers(116)] eglSetDamageRegionKHR failed: eglGetError: EGL_BAD_ACCESS
</pre>

HidenoriMatsubayashi, 2023-03-06:
I found https://community.nxp.com/t5/i-MX-Processors/Issues-with-EGL-in-iMX8MP/m-p/1572714#M199042. What version of BSP are you using?

makotosato-at, 2023-03-06:
Thank you for your reply.

I am using BSP version "hardknott-5.10.72-2.2.0".
But, the latest BSP version "kirkstone-5.15.71-2.2.0" has the same problem.

HidenoriMatsubayashi, 2023-03-06:
Hmm, I'm not sure offhand.

HidenoriMatsubayashi, 2023-03-06:
I fixed a FBO issue https://github.com/sony/flutter-embedded-linux/pull/333. Does it make the issue resolve?

makotosato-at, 2023-03-07:
Thank you very much.
"EGL_BAD_ACCESS" no longer occurs.

But, I now get flickering in my application.
(It works fine in version 3.3.10.)

HidenoriMatsubayashi, 2023-03-07:
Closing the issue.

> But, I now get flickering in my application.
Could you create a new issue? It would be great you attach a video I can watch that symptom.

closed 2023-03-07 by HidenoriMatsubayashi