sony/flutter-embedded-linux #338: Sway segfault in flutter::ELinuxEGLSurface::MakeCurrent()

alexandreMarquis, 2023-04-26:
Hi,
I am using an i.MX8MP board and Sway.
When I use Weston, I can launch the app gallery, but when I use Sway, I get a segmentation fault.
Using GDB, I can trace back to the following:
```
Starting program: /home/user/Gallery/gallery -b .
...
Thread 6 "io.flutter.io" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xffffdd9ee0c0 (LWP 1083)]
0x0000fffff7a87a84 in wl_proxy_set_queue () from /usr/lib/libwayland-client.so.0
#0  0x0000fffff7a87a84 in wl_proxy_set_queue () from /usr/lib/libwayland-client.so.0
#1  0x0000fffff79fc664 in ?? () from /usr/lib/libEGL.so.1
#2  0x0000fffff79ed774 in ?? () from /usr/lib/libEGL.so.1
#3  0x0000fffff7f6c9b4 in flutter::ELinuxEGLSurface::MakeCurrent() const () from /home/user/Gallery/lib/libflutter_elinux_wayland.so
#4  0x0000fffff704929c in ?? () from /home/user/Gallery/lib/libflutter_engine.so
#5  0x0000fffff758b218 in ?? () from /home/user/Gallery/lib/libflutter_engine.so
#6  0x0000fffff707828c in ?? () from /home/user/Gallery/lib/libflutter_engine.so
#7  0x0000fffff707e3cc in ?? () from /home/user/Gallery/lib/libflutter_engine.so
#8  0x0000fffff707813c in ?? () from /home/user/Gallery/lib/libflutter_engine.so
#9  0x0000fffff707dc48 in ?? () from /home/user/Gallery/lib/libflutter_engine.so
#10 0x0000fffff7b902a8 in start_thread (arg=0xffffffffddf7) at pthread_create.c:442
#11 0x0000fffff7bf7bdc in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:79

```

Also the following command yield : 

```
>FLUTTER_LOG_LEVELS=TRACE ./gallery -b .
[TRACE][operator()(171)] presentation info: clk_id = 1
[INFO][operator()(573)] Display output info: width = 1920, height = 1080, refresh = 60000
[INFO][operator()(606)] Display scale for output(10): 1
[TRACE][CreateRenderSurface(1148)] Created the Wayland surface: 1280x720

```

HidenoriMatsubayashi, 2023-04-26:
Thanks for the bug report. 

Can this bug be reproducible on desktop Linux PC (e.g. Ubuntu 22)?

alexandreMarquis, 2023-04-28:
im going to try to setup a vm ubuntu jammy jellyfish arm64. Ill get back to you soon!

alexandreMarquis, 2023-05-10:
Im back with some news, I have built a VM with fedora aarch64, installed sway + the flutter app and they were working fine. 
Turn out the driver imx-gpu-viv form NXP (libEGL.so.1) doesnt fully support wlroot .  

You can have more info: [https://community.nxp.com/t5/i-MX-Processors/GLFW-Vulkan-applications-crash-on-imx8mp-with-wlroots-based-WM/td-p/1572600](https://community.nxp.com/t5/i-MX-Processors/GLFW-Vulkan-applications-crash-on-imx8mp-with-wlroots-based-WM/td-p/1572600 )

closed 2023-05-10 by alexandreMarquis

t123yh, 2023-06-22:
You may try to use the open-source mesa etnaviv driver instead.