sony/flutter-embedded-linux #340: Strange Performance Issue, Possibly Related To Skia?

Ariphonia, 2023-06-07:
Hi,
I am running the flutter engine in release (aot) mode with the wayland embedder backend on a proprietary armv7hf yocto platform and am experiencing a strange performance issue with the `flutter create sample` app where even though it's running in aot mode it's very unresponsive on startup but becomes perfectly responsive after the button has been pressed a large number of times (40-100). The only thing I've found in the flutter docs that might fit this behavior is skia compilation shader jank. I also ran the app on the debug mode flutter engine on the device to get tracing through the dart vm observatory and it seems to fit the description of skia shader compilation jank. Below I have attached a screen capture of the timeline.

[image]

For context, since flutter-elinux currently doesn't support armv7 I am performing aot cross-compilation manually via the manual cross-build instructions provided in the wiki with some adjustments to account for the platform. I have not yet managed to get flutter devtools running on the device. I would like to know if you have any ideas what might be causing this issue and how I could start going about making a fix.

HidenoriMatsubayashi, 2023-06-07:
Thank you for sending this report. I think this is well-known issue called [Shader compilation jank](https://docs.flutter.dev/perf/shader). The only way at this time is to use SkSL warmup.

Ariphonia, 2023-06-08:
I see, thank you. In that case probably going to hold off for now and revisit when impeller opengles becomes available for embedded linux.

0xkelvin, 2024-02-14:
> I see, thank you. In that case probably going to hold off for now and revisit when impeller opengles becomes available for embedded linux.

sorry, might not related topic, but might i ask how do you measure/profiling performance flutter on embedded devices ?