sony/flutter-embedded-linux #341: Can't interact with the application when using RDP

topolittle, 2023-06-08:
When connecting to the device remotely using the remote desktop protocol, it's possible to launch the Flutter application using the embedder, but it's not possible to interact with it once it's launched.
### Environment
Hardware: i.MX8M-Plus EVK (Eval board)
OS version: NXP-5.15.71_2.2.0 (Yocto/Kirkstone)
Window manager:  Weston 10.0.1
Virtual keyboard: Maliit 2.3.0
RDP server: Built-in Weston using rdp-backend.so
RDP client: Remmina 1.4.30 on a Fedora 38 machine
Embedder version (flutter-client): 3dedfc0b674c1c5b2e9a4ddc0a86542cfd70ba56
Backend: Wayland

### Step to reproduce:
1. Enable RDP in Weston, see [weston-ini man page](https://man.archlinux.org/man/weston.ini.5.en) and [weston-rdp man page](https://www.mankier.com/7/weston-rdp)
2. Restart the Weston service
3. Connect to the device over RDP
4. Launch a terminal (weston-terminal)
5. Launch the Flutter app using the embedder: `/home/root/flutter-gallery-elinux/gallery --bundle=/home/root/flutter-gallery-elinux --onscreen-keyboard --window-decoration --width=450 --height=650`

### Expected result
The application is launched and we can interact with it

### Actual result
The application is launched but it's not possible to interact with it, the app doesn't seems to accept any input from RDP. However, it works locally on the device using a keyboard, mouse or touch-screen, event when launched from RDP.

### References
**RDP section in weston.ini:**
```sh
[screen-share]
command=/usr/bin/weston --backend=rdp-backend.so --shell=fullscreen-shell.so --no-clients-resize --rdp-tls-cert=/etc/freerdp/keys/server.crt --rdp-tls-key=/etc/freerdp/keys/server.key
start-on-startup=true
```
### Note
The issue is not present when launching the same application compiled to target the Linux desktop platform (Google's embedder using GTK). It's is working fine locally as well as remotely using RDP in that case.

topolittle, 2023-06-13:
> I have never used RDP in Weston. How about other wayland apps? Do they work fine?

Yes, they do. I tried weston-editor, Gimp and Wireshark. Even a flutter Desktop app (gtk) works fine.

HidenoriMatsubayashi, 2023-08-11:
Sorry for no updates, but I'll investigate this problem next week.

topolittle, 2023-08-11:
Thanks you. It's still an issue on my side.

HidenoriMatsubayashi, 2023-08-14:
Sorry, I tried to RDP on weston, but I couldn't setup/use this feature on Raspberry Pi4 in the first place. Can you please get trace level logs instead?

```
$ cmake -DUSER_PROJECT_PATH=examples/flutter-wayland-client -DCMAKE_BUILD_TYPE=Debug ..
$ FLUTTER_LOG_LEVELS=TRACE  ./flutter-client -b <path_to_bundle>
```

topolittle, 2023-09-05:
> So, can you please get trace level logs instead? Or can I try RDP on Ubuntu desktop PC?
> 
> ```
> $ cmake -DUSER_PROJECT_PATH=examples/flutter-wayland-client -DCMAKE_BUILD_TYPE=Debug ..
> $ FLUTTER_LOG_LEVELS=TRACE  ./flutter-client -b <path_to_bundle>
> ```

Sorry for my late reply, I was in vacation. I will try to get the traces soon and let you know.

mmerah, 2024-02-08:
Hello,

I have had a similar problem, which I believe has the same origin. Essentially, I was trying to open a VNC session to interact remotely with a flutter application. However, I've noticed that once a VNC session is opened, touch input on the target does not work anymore and even if the VNC session is closed the touch inputs are still not functioning.

I use Torizon OS on an iMX8M Plus, so I have to work through containers. However, it should be reproducible with similar steps on any platform. I have an image with flutter-elinux built into it:
```
FROM --platform=linux/arm64/v8 torizon/weston-vivante:3 AS runtime
ARG DEBIAN_FRONTEND=noninteractive

# Build environment dependencies for flutter-elinux
RUN apt-get -y update && apt-get install -y \
    build-essential clang cmake \
    curl git pkg-config \
    ninja-build unzip wget \
    libgtk2.0-dev libglib2.0-dev libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev \
    libgl-dev libegl-dev libegl1 libegl1-mesa-dev libgtk-3-dev \
    libxkbcommon-dev libgles2-mesa-dev \
    libwayland-dev wayland-protocols \
    libavfilter8 libopencl1 \
    gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly \
	gstreamer1.0-libav gstreamer1.0-tools gstreamer1.0-x gstreamer1.0-alsa \
	gstreamer1.0-gl gstreamer1.0-gtk3 gstreamer1.0-pulseaudio v4l-utils \
    locales

# Set the locale
RUN sed -i '/en_US.UTF-8/s/^# //g' /etc/locale.gen && \
    locale-gen
ENV LANG en_US.UTF-8
ENV LANGUAGE en_US:en
ENV LC_ALL en_US.UTF-8

# Clone Flutter 3.16.0
RUN git clone --branch 3.16.0 https://github.com/sony/flutter-elinux.git

# Add flutter-elinux to the path
ENV PATH="${PATH}:/home/torizon/flutter-elinux/bin"
```

I run it with a weston image with ENABLE_VNC=1 on the target:
```
docker run -e ACCEPT_FSL_EULA=1 -e ENABLE_VNC=1 -d --rm --name=weston --net=host --cap-add CAP_SYS_TTY_CONFIG -v /dev:/dev -v /tmp:/tmp -v /run/udev/:/run/udev/ --device-cgroup-rule='c 4:* rmw' --device-cgroup-rule='c 13:* rmw' --device-cgroup-rule='c 199:* rmw' --device-cgroup-rule='c 226:* rmw' torizon/weston-vivante:$CT_TAG_WESTON_VIVANTE --developer --tty=/dev/tty7
```
I called the flutter-elinux image `flutter-client`:
```
docker run -it --rm --name=flutter -v /tmp:/tmp -v /dev/dri:/dev/dri -v /dev/galcore:/dev/galcore --device-cgroup-rule='c 199:* rmw' --device-cgroup-rule='c 226:* rmw' nddrnd/flutter-client:arm64
```
Then, I create a sample app, run it to try, build flutter-client with the debug engine .so from the sample and open a VNC session:
```
flutter-elinux create sample
cd sample
flutter-elinux run -d elinux-wayland
cd ..
git clone https://github.com/sony/flutter-embedded-linux.git
mkdir build
cd build
cp ../../sample/build/elinux/arm64/debug/bundle/lib/* .
cmake -DUSER_PROJECT_PATH=examples/flutter-wayland-client -DCMAKE_BUILD_TYPE=Debug ..
cmake --build .
FLUTTER_LOG_LEVELS=TRACE ./flutter-client -f -b ../../sample/build/elinux/arm64/debug/bundle
```
1. I open a VNC session:
```
[TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
[TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
[TRACE][elinux_window_wayland.cc(287)] wl_seat_listener.capabilities
[TRACE][elinux_window_wayland.cc(316)] wl_seat_listener.name
[TRACE][elinux_window_wayland.cc(70)] xdg_wm_base_listener.ping
[TRACE][elinux_window_wayland.cc(327)] wl_pointer_listener.enter
[TRACE][elinux_window_wayland.cc(208)] wp_presentation_feedback_listener.sync_output
[TRACE][elinux_window_wayland.cc(221)] wp_presentation_feedback_listener.presented
[... 19 lines cut ...]
[TRACE][elinux_window_wayland.cc(60)] wl_registry_listener.global_remove
```
3. I try the touchscreen on my device, nothing happens anymore (no logs):
4. I reopen a VNC session
 ```
[TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
[TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
[TRACE][elinux_window_wayland.cc(287)] wl_seat_listener.capabilities
[TRACE][elinux_window_wayland.cc(316)] wl_seat_listener.name
```
5. I click with the mouse in the VNC window and then close it:
 ```
[TRACE][elinux_window_wayland.cc(70)] xdg_wm_base_listener.ping
[TRACE][elinux_window_wayland.cc(70)] xdg_wm_base_listener.ping
[TRACE][elinux_window_wayland.cc(70)] xdg_wm_base_listener.ping
[TRACE][elinux_window_wayland.cc(70)] xdg_wm_base_listener.ping
[TRACE][elinux_window_wayland.cc(70)] xdg_wm_base_listener.ping
[TRACE][elinux_window_wayland.cc(70)] xdg_wm_base_listener.ping
[TRACE][elinux_window_wayland.cc(60)] wl_registry_listener.global_remove
[TRACE][elinux_window_wayland.cc(60)] wl_registry_listener.global_remove
```

For me, this seems similar to a discussion I saw on the Torizon forums (https://community.toradex.com/t/weston-rdp-vnc-no-mouse-input/19841/11) and a patch was mentioned which consists in supporting multiple wl_seats (https://github.com/Igalia/cog/commit/8e8b8218a7133d410ab657ee69b27fb029d85be7). The related issue (https://github.com/Igalia/cog/issues/214) mentions that "Wayland platform module only picks input events from the first event sources (keyboard, touch, pointer, etc.) advertised by the compositor". What do you think?

Best regards,

omarhogni, 2024-05-23:
Came here from #409 seeing the same issue. Here are TRACE logs of a release build when this occurs.
```
May 23 09:28:35 intel-corei7-64 systemd[1]: Started Start frontend on startup.
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(53)] wl_registry_listener.global
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(1351)] Created the Wayland surface: 1280x720
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(194)] wp_presentation_listener.clock_id
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(198)] clk_id = 1
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(287)] wl_seat_listener.capabilities
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(316)] wl_seat_listener.name
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(630)] wl_output_listener.geometry
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(688)] wl_output_listener.scale
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [INFO][elinux_window_wayland.cc(694)] Display scale for output(10): 1
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(638)] wl_output_listener.mode
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [INFO][elinux_window_wayland.cc(646)] Display output info: width = 1280, height = 800, refresh = 59810
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(685)] wl_output_listener.done
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(630)] wl_output_listener.geometry
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(688)] wl_output_listener.scale
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [INFO][elinux_window_wayland.cc(694)] Display scale for output(11): 1
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(638)] wl_output_listener.mode
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [INFO][elinux_window_wayland.cc(646)] Display output info: width = 1280, height = 800, refresh = 59810
May 23 09:28:35 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(685)] wl_output_listener.done
[... cut ...]
May 23 09:29:01 intel-corei7-64 flutter-client[24184]: [TRACE][elinux_window_wayland.cc(327)] wl_pointer_listener.enter
```
Looking at the source it looks like wl_touch is set to null for the second capabilities line. See https://github.com/sony/flutter-embedded-linux/blob/97438621dbd6b5899859c052ad2ef55843373de4/src/flutter/shell/platform/linux_embedded/window/elinux_window_wayland.cc#L305
If you would like an example configuration for weston to get vnc up and running I can assist with that also.

omarhogni, 2024-05-23:
I created a draft PR https://github.com/sony/flutter-embedded-linux/pull/417 for handling multiple seats.

closed 2024-05-23 by HidenoriMatsubayashi