sony/flutter-embedded-linux #342: DRM Display Blanking

edrose-tewke, 2023-06-15:
I'm just wondering whether there is any way of disabling the display when using the DRM embedder? I guess in DRM terms, that means disabling the CRTC, which will have the knock-on effect of disabling the display further down the chain.

If not, it's a problem I'm prepared to tackle myself (and contribute, if it's a useful feature). However I'm not an expert in flutter. Do you have any recommendation around how to signal down from the flutter app to the embedded to indicate that the display must be turned off? Could it be as simple as just detecting a frame that contains only black pixels?

HidenoriMatsubayashi, 2023-06-15:
> I'm just wondering whether there is any way of disabling the display when using the DRM embedder?

No, there isn't.

> Do you have any recommendation around how to signal down from the flutter app to the embedded to indicate that the display must be turned off? Could it be as simple as just detecting a frame that contains only black pixels?

If my understanding is correct, official flutter doesn't support screen controls (on/off) including platform specific functions as it's just UI framework. Instead, flutter provides plugin interface.

In this case, we should use method channel (or event channel) APIs.
e.g.
https://stackoverflow.com/questions/61889443/flutter-app-ability-to-turn-off-the-screen

Thor-x86, 2024-05-29:
As this issue still opened, I'm going to answer anyway.

> Do you have any recommendation around how to signal down from the flutter app to the embedded to indicate that the display must be turned off? Could it be as simple as just detecting a frame that contains only black pixels?

Thankfully, Flutter on Linux is now supporting [Unix Socket][1]. Thus, you can create a daemon program that listen signal from Flutter via Unix Socket then [switch off the display backlight][2].

[1]: https://stackoverflow.com/a/62006070/3730834
[2]: https://superuser.com/a/1095580/972970