sony/flutter-embedded-linux #345: flutter_elinux.h not C importable

lhoward, 2023-07-16:
`flutter_elinux.h` appears to be C-compatible by virtue of its `extern "C"` guard but it includes some definitions with reference parameters:

```c++
FLUTTER_EXPORT FlutterDesktopViewControllerRef
FlutterDesktopViewControllerCreate(
    const FlutterDesktopViewProperties& view_properties,
    FlutterDesktopEngineRef engine);

FLUTTER_EXPORT FlutterDesktopEngineRef FlutterDesktopEngineCreate(
    const FlutterDesktopEngineProperties& engine_properties);
```

This causes problems with languages such as Swift that do not support C++ interoperability (at least, in shipping versions).

Upstream raised and fixed this in flutter/flutter#75465.

lhoward, 2023-07-16:
Also would be nice to import `FlutterDesktopMessengerAddRef` and `FlutterDesktopMessengerRelease` if possible.

lhoward, 2023-07-16:
Should be fixed in `https://github.com/PADL/flutter-embedded-linux/tree/lhoward/345` (didn't create a PR, per README).

HidenoriMatsubayashi, 2023-07-16:
Thank you. Can you please create a PR with Developer’s Certificate of Origin (DCO) in your commit message? Also, please update (AUTHORS)[https://github.com/sony/flutter-embedded-linux/blob/master/AUTHORS] too.

HidenoriMatsubayashi, 2023-07-17:
> Also would be nice to import FlutterDesktopMessengerAddRef and FlutterDesktopMessengerRelease if possible.

Remaining task.

HidenoriMatsubayashi, 2023-07-17:
Okay, thank you.

HidenoriMatsubayashi, 2023-07-17:
Closing. Thank you so much, Ihoward!

closed 2023-07-17 by HidenoriMatsubayashi

lhoward, 2023-07-17:
Thank you for merging so quickly. I've tested the latest `main` with my [Swift wrapper](https://github.com/PADL/FlutterSwift) and it is working :)