sony/flutter-embedded-linux #382: Support DRM-GBM Backend for IMX GPU Vivante 6 Driver

nguyenlkdn, 2023-08-26:
> `| /mnt/data/yocto/dvt2/yocto/ion-build/tmp/work/cortexa35-mx8-poky-linux/flutter-drm-gbm-backend/1.0-r0/git/src/flutter/shell/platform/linux_embedded/window/native_window_drm_gbm.cc:69:25: error: static_cast from 'EGLNativeWindowType' (aka 'wl_egl_window *') to 'gbm_surface *', which are not related by inheritance, is not allowed
> |     gbm_surface_destroy(static_cast<gbm_surface*>(window_));
> |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> | /mnt/data/yocto/dvt2/yocto/ion-build/tmp/work/cortexa35-mx8-poky-linux/flutter-drm-gbm-backend/1.0-r0/recipe-sysroot/usr/include/gbm.h:47:8: note: 'gbm_surface' is incomplete
> | struct gbm_surface;`

- imx-gpu-viv-6.4.3.p1.2-aarch64 don't expose header type of gbm_surface so we can't use static_cast anymore.
- mesa had expose that struct but it is already removed from 
https://github.com/Freescale/meta-freescale/blob/master/recipes-graphics/mesa/mesa_%25.bbappend#L4

- Reference for updating:
https://github.com/glmark2/glmark2/blob/master/src/native-state-gbm.cpp

HidenoriMatsubayashi, 2023-08-26:
I'm not sure if this is an issue. In the first place, does imx-gpu-viv-6.4.3.p1.2-aarch64 (NXP, i.mx 8M?) support GBM bacnkend?

nguyenlkdn, 2023-08-27:
@HidenoriMatsubayashi thank for your comment
My testing environment is for i.MX 8QXP

> BB_VERSION           = "1.48.0"
> BUILD_SYS            = "x86_64-linux"
> NATIVELSBSTRING      = "ubuntu-22.04"
> TARGET_SYS           = "aarch64-poky-linux"
> MACHINE              = "gridania-evt2-soc"
> DISTRO               = "imx-wayland"
> DISTRO_VERSION       = "5.10-gatesgarth"
> TUNE_FEATURES        = "aarch64 armv8a crc cortexa35 crypto"
> TARGET_FPU           = ""
> meta
> meta-poky            = "HEAD:13143ea85a1ab7703825c0673128c05845b96cb5"
> meta-oe
> meta-multimedia
> meta-python          = "HEAD:ac4ccd2fbbb599d75ca4051911fcbaca39dbe6d7"
> meta-freescale       = "gatesgarth:c0c3b6108837ff232831b67c9ca87fb5b891182a"
> meta-freescale-3rdparty = "gatesgarth:4297672fe35c2c24350f0eaba3d909f5a2dcee4c"
> meta-freescale-distro = "gatesgarth:50eb2b32e7702bc435049bfe0a98fc65c864c106"
> meta-bsp
> meta-sdk
> meta-ml              = "gatesgarth-5.10.9-1.0.0:83d7642fe53f1cd3871cf2b2692b61459237cd64"
> meta-browser         = "HEAD:ee3be3b5986a4aa0e73df2204a625ae1fe5df37e"
> meta-rust            = "master:973180c3d71350c14cb8bfb3f2ade8be330eb727"
> meta-clang           = "HEAD:61faae011fb95712064f2c58abe6293f0daeeab5"
> meta-gnome
> meta-networking
> meta-filesystems     = "HEAD:ac4ccd2fbbb599d75ca4051911fcbaca39dbe6d7"
> meta-qt5             = "HEAD:8d5672cc6ca327576a814d35dfb5d59ab24043cb"
> meta-python2         = "HEAD:c43c29e57f16af4e77441b201855321fbd546661"
> meta-flutter         = "main:9ad01dbcaec5000e41b3b6a453e5aaa39060a8d7"

I think it should be able to support, reference to glmark2-es2-drm application. 
>         libgbm.so.1 => /usr/lib/libgbm.so.1 (0x0000ffff88de5000)
>         libgbm_viv.so => /usr/lib/libgbm_viv.so (0x0000ffff88977000)

# ldd /usr/bin/glmark2-es2-drm
>         linux-vdso.so.1 (0x0000ffff88ff9000)
>         libjpeg.so.62 => /usr/lib/libjpeg.so.62 (0x0000ffff88e9a000)
>         libdl.so.2 => /lib/libdl.so.2 (0x0000ffff88e86000)
>         libpng16.so.16 => /usr/lib/libpng16.so.16 (0x0000ffff88e43000)
>         libz.so.1 => /lib/libz.so.1 (0x0000ffff88e1c000)
>         libdrm.so.2 => /usr/lib/libdrm.so.2 (0x0000ffff88df9000)
>         libgbm.so.1 => /usr/lib/libgbm.so.1 (0x0000ffff88de5000)
>         libudev.so.1 => /lib/libudev.so.1 (0x0000ffff88da1000)
>         libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x0000ffff88bca000)
>         libm.so.6 => /lib/libm.so.6 (0x0000ffff88b20000)
>         libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x0000ffff88afb000)
>         libc.so.6 => /lib/libc.so.6 (0x0000ffff8898c000)
>         /lib/ld-linux-aarch64.so.1 (0x0000ffff88fc8000)
>         libgbm_viv.so => /usr/lib/libgbm_viv.so (0x0000ffff88977000)
>         librt.so.1 => /lib/librt.so.1 (0x0000ffff8895d000)
>         libpthread.so.0 => /lib/libpthread.so.0 (0x0000ffff8892d000)
>         libGAL.so => /usr/lib/libGAL.so (0x0000ffff8876b000)
>         libwayland-server.so.0 => /usr/lib/libwayland-server.so.0 (0x0000ffff88746000)
>         libffi.so.7 => /usr/lib/libffi.so.7 (0x0000ffff8872d000)

/usr/bin/glmark2-es2-drm -b buffer
> Aug 27 01:17:24 gridania-evt2-soc glmark2-es2-drm[226]: =======================================================
> Aug 27 01:17:24 gridania-evt2-soc glmark2-es2-drm[226]:     glmark2 2017.07
> Aug 27 01:17:24 gridania-evt2-soc glmark2-es2-drm[226]: =======================================================
> Aug 27 01:17:24 gridania-evt2-soc glmark2-es2-drm[226]:     OpenGL Information
> Aug 27 01:17:24 gridania-evt2-soc glmark2-es2-drm[226]:     GL_VENDOR:     Vivante Corporation
> Aug 27 01:17:24 gridania-evt2-soc glmark2-es2-drm[226]:     GL_RENDERER:   Vivante GC7000L
> Aug 27 01:17:24 gridania-evt2-soc glmark2-es2-drm[226]:     GL_VERSION:    OpenGL ES 3.1 V6.4.3.p1.305572
> Aug 27 01:17:24 gridania-evt2-soc glmark2-es2-drm[226]: =======================================================
> Aug 27 01:17:29 gridania-evt2-soc glmark2-es2-drm[226]: [buffer] <default>:
> Aug 27 01:17:29 gridania-evt2-soc systemd[1]: psplash-start.service: Succeeded.

/usr/bin/glmark2-es2-drm

> Aug 27 01:18:59 gridania-evt2-soc glmark2-es2-drm[226]: =======================================================
> Aug 27 01:18:59 gridania-evt2-soc glmark2-es2-drm[226]:     glmark2 2017.07
> Aug 27 01:18:59 gridania-evt2-soc glmark2-es2-drm[226]: =======================================================
> Aug 27 01:18:59 gridania-evt2-soc glmark2-es2-drm[226]:     OpenGL Information
> Aug 27 01:18:59 gridania-evt2-soc glmark2-es2-drm[226]:     GL_VENDOR:     Vivante Corporation
> Aug 27 01:18:59 gridania-evt2-soc glmark2-es2-drm[226]:     GL_RENDERER:   Vivante GC7000L
> Aug 27 01:18:59 gridania-evt2-soc glmark2-es2-drm[226]:     GL_VERSION:    OpenGL ES 3.1 V6.4.3.p1.305572
> Aug 27 01:18:59 gridania-evt2-soc glmark2-es2-drm[226]: =======================================================
> Aug 27 01:19:03 gridania-evt2-soc glmark2-es2-drm[226]: [build] use-vbo=false:
> Aug 27 01:19:03 gridania-evt2-soc systemd[1]: psplash-start.service: Succeeded.

HidenoriMatsubayashi, 2023-08-27:
> error: static_cast from 'EGLNativeWindowType' (aka 'wl_egl_window *') to 'gbm_surface *', which are not related by inheritance, is not allowed | [...]

I'm not sure why this error occurred. Because if you are using GBM, `__GBM__` must be defined. But it looks like `WL_EGL_PLATFORM` (wayland) is used.

eglplatform.h:
```
#elif defined(WL_EGL_PLATFORM)

typedef struct wl_display     *EGLNativeDisplayType;
typedef struct wl_egl_pixmap  *EGLNativePixmapType;
typedef struct wl_egl_window  *EGLNativeWindowType;

#elif defined(__GBM__)

typedef struct gbm_device  *EGLNativeDisplayType;
typedef struct gbm_bo      *EGLNativePixmapType;
typedef void               *EGLNativeWindowType;
```

closed 2023-09-05 by HidenoriMatsubayashi