sony/flutter-embedded-linux #383: Atomic Modeset fails

hundeboll, 2023-08-30:
Hi,

I'm trying to run flutter-embedded-linux on a buildroot based linux using the eglstream backend for an nvidia card. I've enabled modesetting on the command-line, but it fails like this:
```
# LD_LIBRARY_PATH=/root/drm/lib/ FLUTTER_DRM_DEVICE=/dev/dri/card0 ./flutter-drm-eglstream-backend -b drm/
[ERROR][native_window_drm_eglstream.cc(132)] Failed to commit an atomic property change request
[ERROR][elinux_window_drm.h(131)] Failed to create the native window (/dev/dri/card0).
Segmentation fault (core dumped)
```

The [dmesg.log](https://github.com/sony/flutter-embedded-linux/files/12476208/dmesg.log) (with `drm.debug=0xff` on the command line) does give a hint about no framebuffer being set.

I am by no means an DRM expert, so any hints would be greatly appreciated.

HidenoriMatsubayashi, 2023-08-30:
> I'm trying to run flutter-embedded-linux on a buildroot based linux using the eglstream backend for an nvidia card.

Specifically, what device are you using? Also, did you run that command with `sudo`?

hundeboll, 2023-08-30:
Running as root, yes. With the suggested changes:
```
# LD_LIBRARY_PATH=/root/drm/lib/ FLUTTER_DRM_DEVICE=/dev/dri/card0 ./flutter-drm-eglstream-backend -b drm/
[ERROR][native_window_drm_eglstream.cc(132)] Failed to commit an atomic property change request: -22
[ERROR][environment_egl_stream.cc(16)] Failed to set extension function pointers
[ERROR][context_egl.cc(71)] Failed to choose EGL surface config: eglGetError: EGL_BAD_DISPLAY
[ERROR][context_egl_stream.cc(20)] Failed to set extension function pointers
Segmentation fault (core dumped)
```

HidenoriMatsubayashi, 2023-08-30:
> [ERROR][native_window_drm_eglstream.cc(132)] Failed to commit an atomic property change request: -22

-22 means:

-EINVAL | if DRM_CLIENT_CAP_ATOMIC is not enabled, the value of flags is illegal, or atomic property IDs in the request are not recognized.
-- | --

BTW, what device are you using?

hundeboll, 2023-08-30:
`NVIDIA GPU GeForce GTS 450 (GF106)`
`NVIDIA GLX Module  390.151`

HidenoriMatsubayashi, 2023-08-30:
I see. I haven't tried this backend on desktop graphics cards.

Sorry, I can't think of anything right away. Need investigation.

hundeboll, 2023-08-30:
No problem. Thanks for a prompt response!

HidenoriMatsubayashi, 2023-08-30:
If you find a bug, please send a pull request :)

kintonfull, 2024-11-22:
I have the same problem on "NVIDIA Jetson Orin NX"
CPU: aarch64
System: Ubuntu 20.04 focal
Kernel: 5.10.192-tegra
Vulkan: 1.3.204
CUDA: 11.4.315
L4T: 35.5.0
Jetpack: 5.1.3