sony/flutter-embedded-linux #384: flutter app dies after updating from 05b0377d52662a33d53370f745e2f127d4ec0248

angelobt, 2023-09-01:
we are using weston compositor with kiosk shell.

We have a flutter application using flutter-wayland-client.

updating from 05b0377d52662a33d53370f745e2f127d4ec0248 to head (1 sept 2023) we run into this error

[FATAL:flutter/shell/common/display_manager.cc(32)] Check failed: displays_.empty().

that happens if an other application (like gstreamer) wants to render something through wayland.

HidenoriMatsubayashi, 2023-09-01:
I feel https://github.com/sony/flutter-embedded-linux/pull/374 is probably the cause.

HidenoriMatsubayashi, 2023-09-01:
This issue shouldn't occur in release flavor.

https://github.com/flutter/engine/blob/main/shell/common/display_manager.cc#L27C15-L27C23
```
void DisplayManager::HandleDisplayUpdates(
    std::vector<std::unique_ptr<Display>> displays) {
  FML_DCHECK(!displays.empty());
  std::scoped_lock lock(displays_mutex_);
  displays_ = std::move(displays);
}

```

angelobt, 2023-09-01:
Thanks for the support. As soon as possible we will try the release flavor

omarhogni, 2023-09-05:
+1 for this issue, same setup weston, kiosk shell

HidenoriMatsubayashi, 2023-09-05:
Can you please share the detailed steps to reproduce this crash?

omarhogni, 2023-09-05:
Switching to 7f83ccb9a9790ab31758e76b3b5498be2af32af7 it looks like the error is gone.

omarhogni, 2023-09-05:
For me this is not a crash but just at startup. But I am consuming this project through [meta-flutter ](https://github.com/meta-flutter/meta-flutter) and using that project to build the engine and do the heavy lifting. then I just start it.

What kind of steps are you looking for?

HidenoriMatsubayashi, 2023-09-05:
What version of flutter/engine are you using?

> What kind of steps are you looking for?

Just how to reproduce this issue / your steps. I haven't used weston + kiosk shell.

HidenoriMatsubayashi, 2023-09-05:
Just in case, `FlutterEngineDisplay` API of flutter/engine's embedder has been supported since flutter 3.13.0. If you use the latest this software, you need to use flutter version 3.13.0 or higher.

omarhogni, 2023-09-05:
I am using 3.7.7 so that might be it.

HidenoriMatsubayashi, 2023-09-05:
Definitely, yes. Please use 3.13.0 or higher. I'll close this issue. Please let me know if you get this error even if you use the correct versions. I'll re-open it.

closed 2023-09-05 by HidenoriMatsubayashi

angelobt, 2023-09-05:
I'm using flutter-engine  3.3.10. So it makes sense