sony/flutter-embedded-linux #387: DRM mode on raspberry pi 4b Error: Couldn't find file "symbols/US" in include paths

talksik, 2023-09-22:
Environment: Raspberry pi 4b running headless mode (Raspbian OS lite 64bit)

## Problem
I installed all dependencies as per [this](https://github.com/sony/flutter-elinux/wiki/flutter-elinux-install).

1. SSH into raspberry pi 4b
2. cd /opt
3. flutter-elinux create helloworld
4. cd helloworld
5. flutter-elinux build elinux --debug --target-backend-type=gbm
6. cd build/.../bundle
7. Run the below command
```
sudo FLUTTER_DRM_DEVICE=/dev/dri/card1 ./helloworld -b /opt/helloworld/build/elinux/arm64/debug/bundle/
xkbcommon: ERROR: Couldn't find file "symbols/US" in include paths
xkbcommon: ERROR: 1 include paths searched:
xkbcommon: ERROR:       /usr/share/X11/xkb
xkbcommon: ERROR: 3 include paths could not be added:
xkbcommon: ERROR:       /root/.config/xkb
xkbcommon: ERROR:       /root/.xkb
xkbcommon: ERROR:       /etc/xkb
xkbcommon: ERROR: Abandoning symbols file "(unnamed)"
xkbcommon: ERROR: Failed to compile xkb_symbols
xkbcommon: ERROR: Failed to compile keymap
Segmentation fault
```

## Solution
I was able to fix this by doing a mere:
`sudo cp /usr/share/X11/xkb/symbols/us /usr/share/X11/xkb/symbols/US`

I am no expert on linux or flutter embedded. For some reason the libxkbcommon package on raspbian has the symbols file for US layout as lowercase 'us'. 

I hope that this helps others out there!

## Question
What part of elinux is setting this include path of "symbols/US"? Or is it a bug in Raspbian?

HidenoriMatsubayashi, 2023-09-23:
I'm not sure. I'm using the latest Raspbian OS (2023-05-03-raspios-bullseye-arm64.img), but it works fine without that error. So, it depends on your locale, keyboard settings and so on.

BTW, you created https://github.com/sony/flutter-elinux/issues/199 before, right? You prefer Qt rather than flutter, but this is an open source software and no one is in charge and there is no commitment. So please don't avoid such kind of issues in the future🙂

talksik, 2023-09-23:
@HidenoriMatsubayashi sorry if [that post](https://github.com/sony/flutter-elinux/issues/199) came across as negative. 

When you say, don't create or put comments, do you mean this post or the linked post?

Anyway, I wanted to just help others out there with this post. The linked post about Qt was also just to help others. If you would like to re-word that post, please do so.

HidenoriMatsubayashi, 2023-09-23:
I mean the lined post. Thanks. Closing this issue.

closed 2023-09-23 by HidenoriMatsubayashi