sony/flutter-embedded-linux #396: xdg_wm_base@11: error 4: xdg_surface buffer (1280 x 720) is larger than the configured fullscreen state (1024 x 600)

makotosato-at, 2023-10-30:
Hello.

I'm using RPi4.
In version 3.13.0(or later) of flutter-elinux,
the following error occurs when I launch an application with `-f` option on weston(resolution is 1024x600).
```
xdg_wm_base@11: error 4: xdg_surface buffer (1280 x 720) is larger than the configured fullscreen state (1024 x 600)
```
Is there any solution to this problem?

It works fine with version 3.10.6.

I attach the logs(WAYLAND_DEBUG=1 and FLUTTER_LOG_LEVELS=TRACE) and screenshot.
[wayland_debug_log.txt](https://github.com/sony/flutter-embedded-linux/files/13203487/wayland_debug_log.txt)
[trace_log.txt](https://github.com/sony/flutter-embedded-linux/files/13203490/trace_log.txt)
[image]

HidenoriMatsubayashi, 2023-10-30:
It looks like https://github.com/sony/flutter-embedded-linux/pull/366 causes this issue.

HidenoriMatsubayashi, 2023-10-30:
https://github.com/sony/flutter-embedded-linux/pull/397 should fix the issue. Thanks.

makotosato-at, 2023-10-31:
Works fine!
Thank you very much!!