sony/flutter-embedded-linux #399: Missing calls to "registerWith()" in generated dart_plugin_registrant.dart

zr6328, 2023-10-31:
1. create a plugin by dartPluginClass
2. create a flutter project
3. compile the app by using cross-build's method
4. open the file .dart_tool/flutter_build/dart_plugin_registrant.dart
5. find the symbol of registerWith in libapp.so

In linux-x64,we can find the symbol of registerWith in libapp.so,but in linux-arm64,we can't find the symbol of registerWith in libapp.so

the plugin's pubspec.yaml look like this:
```dart
  plugin:
      implements: plugin_platform_template_interface
      platforms:
        macos:
          dartPluginClass: FlutterMacPlugin
```

the dart_plugin_registrant.dart look like this:
```dart
@pragma('vm:entry-point')
class _PluginRegistrant {
  @pragma('vm:entry-point')
  static void register() {
    print('zrzrzrzr register function called!!!!');
    if (Platform.isAndroid) {
      print("zrzrzrzr platform is android");
    } else if (Platform.isIOS) {
      print("zrzrzrzr platform is ios");
    } else if (Platform.isLinux) {
      print("zrzrzrzr platform is Linux");
      try {
        FlutterLinuxPlugin.registerWith();
      } catch (err) {
        print(
          '`flutter_linux_plugin` threw an error: $err. '
          'The app may not function as expected until you remove this plugin from pubspec.yaml'
        );
      }

    } else if (Platform.isMacOS) {
      print("zrzrzrzr platform is Macos");
      try {
        FlutterMacPlugin.registerWith();
      } catch (err) {
        print(
          '`flutter_mac_plugin` threw an error: $err. '
          'The app may not function as expected until you remove this plugin from pubspec.yaml'
        );
      }

    } else if (Platform.isWindows) {
      print('zrzrzrzr platform is Windows');
    }
  }
}
```

HidenoriMatsubayashi, 2023-10-31:
I believe this is not a bug of this software. Please see https://github.com/flutter/flutter/issues/113720. Thanks.

zr6328, 2023-11-01:
> I believe this is not a bug of this software. Please see [flutter/flutter#113720](https://github.com/flutter/flutter/issues/113720). Thanks.

no, this method can't help me.

HidenoriMatsubayashi, 2023-11-01:
What version of flutter-elinux are you using? Also build commands.

HidenoriMatsubayashi, 2023-11-01:
+ what is the build target?

zr6328, 2023-11-01:
> What version of flutter-elinux are you using? Also build commands.

this is the build commands
$ flutter build bundle --asset-dir=${RESULT_DIR}/${BUILD_MODE}/bundle/data/flutter_assets
$ cp ${FLUTTER_SDK}/bin/cache/artifacts/engine/linux-x64/icudtl.dat \
   ${RESULT_DIR}/${BUILD_MODE}/bundle/data/

# Build kernel_snapshot.
$ ${FLUTTER_SDK}/bin/cache/dart-sdk/bin/dart \
  --verbose \
  --disable-dart-dev ${FLUTTER_SDK}/bin/cache/artifacts/engine/linux-x64/frontend_server.dart.snapshot \
  --sdk-root ${FLUTTER_SDK}/bin/cache/artifacts/engine/common/flutter_patched_sdk_product/ \
  --target=flutter \
  --no-print-incremental-dependencies \
  -Ddart.vm.profile=false \
  -Ddart.vm.product=true \
  --aot \
  --tfa \
  --packages .dart_tool/package_config.json \
  --output-dill .dart_tool/flutter_build/flutter-embedded-linux/app.dill \
  --depfile .dart_tool/flutter_build/flutter-embedded-linux/kernel_snapshot.d \
  package:${APP_PACKAGE_NAME}/main.dart

# Build AOT image.
$ ${FLUTTER_SDK}/bin/cache/artifacts/engine/linux-arm64-release/clang_x64/gen_snapshot \
  --deterministic \
  --snapshot_kind=app-aot-elf \
  --elf=.dart_tool/flutter_build/flutter-embedded-linux/libapp.so \
  --strip \
  .dart_tool/flutter_build/flutter-embedded-linux/app.dill

$ cp .dart_tool/flutter_build/flutter-embedded-linux/libapp.so ${RESULT_DIR}/${BUILD_MODE}/bundle/lib/

zr6328, 2023-11-01:
> * what is the build target?

Linux-x64 as the host, and Linux-arm64 as the target.

HidenoriMatsubayashi, 2023-11-01:
Did you try [flutter-elinux](https://github.com/sony/flutter-elinux) instead of the official tool?

zr6328, 2023-11-01:
> That is the official flutter tool. Did you try [flutter-elinux](https://github.com/sony/flutter-elinux) instead of it? I recommend you to use flutter-elinux than manual commands with the flutter official tool.

ok, i will try it.

zr6328, 2023-11-01:
> That is the official flutter tool. Did you try [flutter-elinux](https://github.com/sony/flutter-elinux) instead of it? I recommend you to use flutter-elinux than manual commands with the flutter official tool.

In flutter-elinux , i can also build the libapp.so?

HidenoriMatsubayashi, 2023-11-01:
I'm not sure what you want to do, but you can build flutter apps with that tool. See Wiki.

zr6328, 2023-11-01:
> I'm not sure what you want to do, but you can build flutter apps with that tool. See Wiki.
I want to build the libapp.so that can run i linux-arm64 in linux-x64 platform. Now, when I use the cross-build method, that can not run in linux-arm64.

zr6328, 2023-11-01:
> I'm not sure what you want to do, but you can build flutter apps with that tool. See Wiki.
> I want to build the libapp.so that can run i linux-arm64 in linux-x64 platform. Now, when I use the cross-build method, that can not run in linux-arm64.

I'm not sure what you want to do, but you can build flutter apps with that tool. See Wiki.
I want to build the libapp.so that can run i linux-arm64 in linux-x64 platform. Now, when I use the cross-build method, that can not run in linux-arm64.

zr6328, 2023-11-01:
> I'm not sure what you want to do, but you can build flutter apps with that tool. See Wiki.

When i run the command of flutter-elinux doctor, i can't find the /cache/artifacts/engine/linux-x64/icudtl.dat in the directory of bin

HidenoriMatsubayashi, 2023-11-01:
You can build flutter apps with flutter-elinux and build artifacts can be found in under build directory. See https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps

HidenoriMatsubayashi, 2023-11-01:
If you use a target for normal Linux not elinux, please ask questions in flutter official repo.

zr6328, 2023-11-01:
> You can build flutter apps with flutter-elinux and build artifacts can be found in under build directory. See https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps

I have clone the flutter-elinux, and run the flutter-elinux doctor, but In the bin's directory, i cant't find the directory of artifacts

HidenoriMatsubayashi, 2023-11-01:
You should read Wiki in flutter-elinux before asking here.