sony/flutter-embedded-linux #403: Disconnecting mouse that is hovering over button on wayland client results in Segmentation fault

fredrmag, 2023-11-28:
## Environment
flutter-embedded-linux: 16aac2ca2ba5003a247e8436b9bd67d2d89224c0
Flutter 3.3.10
Framework • revision 135454af32 • 2022-12-15 07:36:55 -0800
Engine • revision 3316dd8728
Tools • Dart 2.18.6 • DevTools 2.15.0
weston 10.0.1

Hardware: IMX8M mini

## Steps to reproduce
1. Build the sample application and place it on the device. 
2. Start application using flutter-client:
```
export XDG_RUNTIME_DIR=/run/user/1000
export WAYLAND_DISPLAY=wayland-0
flutter-client -b /usr/share/sample -f
```
3. Connect USB mouse
4. Hover over (+) button so that mouse pointer changes icon from "basic" to "click"
5. Disconnect USB mouse.

These steps results in `Segmentation Fault`.

## Debug information
Running the `flutter-client` with `FLUTTER_LOG_LEVELS=TRACE` and doing the above steps, results in the following:
```
FLUTTER_LOG_LEVELS=TRACE flutter-client -b /usr/share/spingui_flutter -f
[TRACE][elinux_window_wayland.cc(171)] presentation info: clk_id = 1
[INFO][elinux_window_wayland.cc(606)] Display scale for output(9): 1
[INFO][elinux_window_wayland.cc(573)] Display output info: width = 1024, height = 600, refresh = 59269
[TRACE][elinux_window_wayland.cc(1148)] Created the Wayland surface: 1024x600
[TRACE][elinux_window_wayland.cc(143)] window entered output 9
Segmentation fault
```
Running  `flutter-client` with the option `-n` does not produce the same error.

Recompiling the `flutter-client` with additional `TRACE` logging, and doing the above steps, results in the following: 
<details><summary>See log</summary>

```
FLUTTER_LOG_LEVELS=TRACE ./flutter-client -b /usr/share/sample -f
[TRACE][elinux_window_wayland.cc(171)] presentation info: clk_id = 1
[INFO][elinux_window_wayland.cc(616)] Display scale for output(9): 1
[INFO][elinux_window_wayland.cc(583)] Display output info: width = 1024, height = 600, refresh = 59269
[TRACE][elinux_window_wayland.cc(1158)] Created the Wayland surface: 1024x600
[TRACE][elinux_window_wayland.cc(143)] window entered output 9
[TRACE][elinux_window_wayland.cc(291)] wl_pointer_listener.enter
[TRACE][elinux_window_wayland.cc(294)] wl_pointer_listener.enter.use_mouse_cursor
[TRACE][elinux_window_wayland.cc(1261)] wl_pointer_: 0xaaaad1c10ba0
[TRACE][elinux_window_wayland.cc(1262)] cursor_info_.cursor_name.empty(): 1
[TRACE][elinux_window_wayland.cc(1263)] cursor_info_.cursor_name:
[TRACE][elinux_window_wayland.cc(1266)] cursor_info_.cursor_name: click
[TRACE][elinux_window_wayland.cc(1279)] GetWlCursor() start
[TRACE][elinux_window_wayland.cc(1281)] GetWlCursor() end
[TRACE][elinux_window_wayland.cc(1288)] UpdateFlutterCursor if(buffer) start
[TRACE][elinux_window_wayland.cc(1289)] cursor_info_.pointer: 0xaaaad1c10ba0
[TRACE][elinux_window_wayland.cc(1290)] cursor_info_.serial: 2015
[TRACE][elinux_window_wayland.cc(1291)] wl_cursor_surface_: 0xaaaad1a38360
[TRACE][elinux_window_wayland.cc(1292)] wl_pointer_set_cursor start
[TRACE][elinux_window_wayland.cc(1296)] wl_pointer_set_cursor end
[TRACE][elinux_window_wayland.cc(1297)] wl_surface_attach start
[TRACE][elinux_window_wayland.cc(1299)] wl_surface_attach end
[TRACE][elinux_window_wayland.cc(1303)] UpdateFlutterCursor if(buffer) end
[TRACE][elinux_window_wayland.cc(314)] wl_pointer_listener.leave
[TRACE][elinux_window_wayland.cc(317)] cursor_info_.cursor_name.clear() start
[TRACE][elinux_window_wayland.cc(321)] cursor_info_.cursor_name.clear() end
[TRACE][elinux_window_wayland.cc(325)] binding_handler_delegate_->OnPointerLeave() start
[TRACE][flutter_elinux_view.cc(397)] event_data.phase == FlutterPointerPhase::kRemove
[TRACE][flutter_elinux_view.cc(399)] ResetMouseState() start
[... 477 lines cut ...]
[TRACE][elinux_window_wayland.cc(1261)] wl_pointer_: 0
[TRACE][elinux_window_wayland.cc(1262)] cursor_info_.cursor_name.empty(): 1
[TRACE][elinux_window_wayland.cc(1263)] cursor_info_.cursor_name:
[TRACE][elinux_window_wayland.cc(1266)] cursor_info_.cursor_name: basic
[TRACE][elinux_window_wayland.cc(1279)] GetWlCursor() start
[TRACE][elinux_window_wayland.cc(1281)] GetWlCursor() end
[TRACE][elinux_window_wayland.cc(1288)] UpdateFlutterCursor if(buffer) start
[TRACE][elinux_window_wayland.cc(1289)] cursor_info_.pointer: 0xaaaad1c10ba0
[TRACE][elinux_window_wayland.cc(1290)] cursor_info_.serial: 2075
[TRACE][elinux_window_wayland.cc(1291)] wl_cursor_surface_: 0xaaaad1a38360
[TRACE][elinux_window_wayland.cc(1292)] wl_pointer_set_cursor start
Segmentation fault
```

</details>

## Interpretation
`UpdateFlutterCursor` in  `elinux_window_wayland.cc` is called after pointer is destroyed.

Adding the following to the top of the `UpdateFlutterCursor` function in `elinux_window_wayland.cc` seems to fix the bug:
```cpp
if (!wl_pointer_){
  return;
}
```

fredrmag, 2023-11-28:
Thanks for fast response. 

I can send a PR to fix the issue. I have no problem with delegating all rights related to this change to Sony. I can add the `--signoff` when I commit to add the DCO in the commit log.

closed 2023-11-28 by HidenoriMatsubayashi

yaowei103, 2025-05-07:
Hi, I still got the error "Segmentation fault", after disconnect the mouse;

Flutter 3.27.1 • channel [user-branch] • unknown source
Framework • revision 17025dd882 (5 个月前) • 2024-12-17 03:23:09 +0900
Engine • revision cb4b5fff73
Tools • Dart 3.6.0 • DevTools 2.40.2

and the same error with 3.24.2

[image]

please have a look, many thanks;

yaowei103, 2025-05-08:
[image]

I checked the source code and found that the root cause is in lines 1523 to 1525. Based on the commit history, it might have been accidentally deleted, but I'm not sure.