sony/flutter-embedded-linux #407: how to measure flutter performance on embedded device

0xkelvin, 2024-02-14:
Hello, 

is there any way to profile or measure : performance, memory consumption... of flutter app on embedded device (my case is nxp imx8x)
i saw https://docs.flutter.dev/tools/devtools/overview quite useful, don't know if any equivalent for embedded device.

thanks

HidenoriMatsubayashi, 2024-02-14:
Sorry, I don't understand what you want to know here...But you can use devtools on embedded devices.

0xkelvin, 2024-02-14:
i basically want to see memory consumption, cpu load when running flutter app on my nxp imx8x.
it is good to know that devtools can use on embedded devices, is they any guideline to use it on embedded devices @HidenoriMatsubayashi ?

thanks

HidenoriMatsubayashi, 2024-02-14:
For memory consumption and cpu load, in general, I think you can use Linux commands like 'top' command.

See https://github.com/sony/flutter-elinux/wiki/Debugging-with-DevTools for the devtools and use flutter-elinux.

HidenoriMatsubayashi, 2024-02-14:
Closing. Thanks.

closed 2024-02-14 by HidenoriMatsubayashi