sony/flutter-embedded-linux #410: Turning the display OFF/ON swaps the width and height of the application

makotosato-at, 2024-03-27:
Hello.

I'm using RPi4 and weston.
I use weston as `transform=rotate-270` in `weston.ini`.

In my environment, the following procedure will swap the width and height of the application.
1. Launch the application -> [application_launch](https://github.com/sony/flutter-embedded-linux/assets/51475851/3d23589a-a9db-4816-9c60-03c91b33ae59)
2. Turn off the display
3. Turn on the display -> [display_turn_off_then_turn_on](https://github.com/sony/flutter-embedded-linux/assets/51475851/d3024efd-23b1-48f3-a2cb-77b2c7fcc3b1)

Is there any solution to this problem?

I attach the trace logs.
[application_launch.txt](https://github.com/sony/flutter-embedded-linux/files/14770954/application_launch.txt)
[display_turn_off_then_turn_on.txt](https://github.com/sony/flutter-embedded-linux/files/14770958/display_turn_off_then_turn_on.txt)

--
flutter-elinux 3.19.3
weston 9.0.0
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr 3 17:24:16 BST 2023 aarch64 GNU/Linux

HidenoriMatsubayashi, 2024-03-28:
Questions.
* Does flutter-elinux with X11 work without the issue?
* Do other wayland apps work?

If both are "Yes", this issue depends on only wayland backend of this software.

HidenoriMatsubayashi, 2024-03-28:
Also, are you using `--fullscreen` option?

makotosato-at, 2024-03-28:
Thank you for reply.

> Does flutter-elinux with X11 work without the issue?

Yes.

> Do other wayland apps work?

Yes. (For example, weston-terminal works fine.)

> Also, are you using --fullscreen option?

Yes. I use --fullscreen (-f) option.

I tried the following patch and it works.
```
diff --git a/src/flutter/shell/platform/linux_embedded/window/elinux_window_wayland.cc b/src/flutter/shell/platform/linux_embedded/window/elinux_window_wayland.cc
index fa147f6..0652954 100644
--- a/src/flutter/shell/platform/linux_embedded/window/elinux_window_wayland.cc
+++ b/src/flutter/shell/platform/linux_embedded/window/elinux_window_wayland.cc
@@ -628,6 +628,8 @@ const wl_output_listener ELinuxWindowWayland::kWlOutputListener = {
                    const char* model,
                    int32_t output_transform) -> void {
       ELINUX_LOG(TRACE) << "wl_output_listener.geometry";
+      auto self = reinterpret_cast<ELinuxWindowWayland*>(data);
+      self->transform_ = output_transform;
     },
     .mode = [](void* data,
                wl_output* wl_output,
@@ -639,7 +641,11 @@ const wl_output_listener ELinuxWindowWayland::kWlOutputListener = {
 
       auto self = reinterpret_cast<ELinuxWindowWayland*>(data);
       if (flags & WL_OUTPUT_MODE_CURRENT) {
-        if (self->current_rotation_ == 90 || self->current_rotation_ == 270) {
+        if (self->current_rotation_ == 90 || self->current_rotation_ == 270
+            || self->transform_ == WL_OUTPUT_TRANSFORM_90
+            || self->transform_ == WL_OUTPUT_TRANSFORM_270
+            || self->transform_ == WL_OUTPUT_TRANSFORM_FLIPPED_90
+            || self->transform_ == WL_OUTPUT_TRANSFORM_FLIPPED_270) {
           std::swap(width, height);
         }
 
diff --git a/src/flutter/shell/platform/linux_embedded/window/elinux_window_wayland.h b/src/flutter/shell/platform/linux_embedded/window/elinux_window_wayland.h
index 2d1343d..dcfe1bb 100644
--- a/src/flutter/shell/platform/linux_embedded/window/elinux_window_wayland.h
+++ b/src/flutter/shell/platform/linux_embedded/window/elinux_window_wayland.h
@@ -153,6 +153,7 @@ class ELinuxWindowWayland : public ELinuxWindow, public WindowBindingHandler {
   bool maximised_;
   uint32_t last_frame_time_;
   bool enable_impeller_ = false;
+  int32_t transform_ = WL_OUTPUT_TRANSFORM_NORMAL;
 
   // Indicates that exists a keyboard show request from Flutter Engine.
   bool is_requested_show_virtual_keyboard_;
```

HidenoriMatsubayashi, 2024-03-28:
Makes sense. Could you please send a PR?

> int32_t transform_ = WL_OUTPUT_TRANSFORM_NORMAL;

Can we use `current_rotation_` var instead of the creation of `transform_`?

makotosato-at, 2024-03-28:
> Can we use current_rotation_ var instead of the creation of transform_?

Unfortunately no.

current_rotation is used in `GetRotationDegree()`
```
uint16_t ELinuxWindowWayland::GetRotationDegree() const {
  return current_rotation_;
}
```

but, the value of `output_transform` is 0 to 7.
```
enum wl_output_transform {
	WL_OUTPUT_TRANSFORM_NORMAL = 0,
	WL_OUTPUT_TRANSFORM_90 = 1,
	WL_OUTPUT_TRANSFORM_180 = 2,
	WL_OUTPUT_TRANSFORM_270 = 3,
	WL_OUTPUT_TRANSFORM_FLIPPED = 4,
	WL_OUTPUT_TRANSFORM_FLIPPED_90 = 5,
	WL_OUTPUT_TRANSFORM_FLIPPED_180 = 6,
	WL_OUTPUT_TRANSFORM_FLIPPED_270 = 7,
};
```

HidenoriMatsubayashi, 2024-03-28:
> but, the value of output_transform is 0 to 7.

Isn't it sufficient to just convert the value when assigning it to `current_rotation_`? Also, it feels incorrect because it would result in a discrepancy with the information on the Flutter Engine side.

HidenoriMatsubayashi, 2024-03-28:
Apologies, it was a misunderstanding on my part. Your original patch is correct because this involves rotation on the Wayland compositor side, so there's no need to convey this information to the Flutter Engine side.

Can you please send the pull request?

makotosato-at, 2024-03-28:
Thank you.
I sent a PR.

closed 2024-03-28 by HidenoriMatsubayashi