sony/flutter-embedded-linux #412: Dart VM: Wrong full snapshot version, expected XXX, found YYY

barlone, 2024-04-01:
I could not use several last releases. Application builds without errors (arm64+X11) but crashes on start with error:

[FATAL:flutter/runtime/dart_vm_initializer.cc(89)] Error while initializing the Dart VM: Wrong full snapshot version, expected 'f71c76320d35b65f1164dbaa6d95fe09' found 'ee1eb666c76a5cb7746faf39d0b97547'

HidenoriMatsubayashi, 2024-04-01:
I don't think this issue depends on this software or flutter-elinux. Can you please try the following?

```
rm -r ${FLUTTER_HOME}/bin/cache/
flutter doctor
```

See also https://stackoverflow.com/questions/53431517/flutter-app-error-while-initializing-the-dart-vm-wrong-full-snapshot-version-e

barlone, 2024-04-02:
rm -r ${FLUTTER_HOME}/bin/cache/