sony/flutter-embedded-linux #413: Incorrect snapshot architecture

lhoward, 2024-04-17:
I'm using the cross-compile script [here](https://github.com/sony/flutter-embedded-linux/wiki/Building-Flutter-apps) to build a release mode shared object for use with my custom runner (I'm not cross-compiling, so the script has been amended appropriately; I don't believe this is relevant).

Since upgrading to 3.19.0 I'm seeing the following error:

```bash
$ ./build-counter.sh 

Snapshot not compatible with the current VM configuration: the snapshot requires 'release no-code_comments no-dwarf_stack_traces_mode lazy_dispatchers no-dedup_instructions no-tsan no-asserts use_field_guards use_osr no-branch_coverage x64 linux no-compressed-pointers null-safety' but the VM has 'release no-code_comments no-dwarf_stack_traces_mode lazy_dispatchers no-dedup_instructions no-tsan no-asserts use_field_guards use_osr no-branch_coverage arm64 linux no-compressed-pointers null-safety'
```

It appears that the cached snapshot artefacts are the wrong architecture:

```bash
$ /opt/flutter-elinux/flutter/bin/cache/artifacts/engine/linux-arm64$ strings *.snapshot|grep no-compressed-pointers
ee1eb666c76a5cb7746faf39d0b97547release no-code_comments no-dwarf_stack_traces_mode lazy_dispatchers no-dedup_instructions no-tsan no-asserts use_field_guards use_osr no-branch_coverage x64 linux no-compressed-pointers null-safety
```

Running `flutter build elinux` does work, however.

HidenoriMatsubayashi, 2024-04-18:
> I'm not cross-compiling

Are you building your flutter app for arm64 targets on arm64 host PC? (= you are doing self-builds on arm64 host?)

lhoward, 2024-04-18:
Yes :)

lhoward, 2024-04-18:
Essentially I want to build just libapp without the runner as I’m using my own runner. But, it’s also fine if this is not possible as I can run the normal build process and extract the artifact.

lhoward, 2024-04-18:
Script here, it’s not cross-compiling, and it used to work.

https://github.com/PADL/FlutterSwift/blob/main/Examples/counter/build-counter.sh

HidenoriMatsubayashi, 2024-04-19:
I think the log messages from `flutter-elinux build elinux --verbose` on arm64 host would would be helpful for updating your build script on an arm64 host. Could you try that?

lhoward, 2024-04-20:
Thank you! I updated the build script accordingly (likely not all changes were required, I presume the important ones were just `dartaotruntime` and correcting the snapshot path).

``diff
diff --git a/build-aot.sh b/build-aot.sh
index 412597a..6dd9128 100755
--- a/build-aot.sh
+++ b/build-aot.sh
@@ -12,7 +12,7 @@ export APP_PACKAGE_NAME=inferno_ui
 
 # The build data.
 export RESULT_DIR=build/elinux/arm64
-export BUILD_MODE=debug
+export BUILD_MODE=release
 
 mkdir -p .dart_tool/flutter_build/flutter-embedded-linux
 mkdir -p ${RESULT_DIR}/${BUILD_MODE}/bundle/lib/
@@ -24,9 +24,9 @@ cp ${FLUTTER_SDK}/bin/cache/artifacts/engine/linux-arm64/icudtl.dat \
    ${RESULT_DIR}/${BUILD_MODE}/bundle/data/
 
 # Build kernel_snapshot.
-${FLUTTER_SDK}/bin/cache/dart-sdk/bin/dart \
+${FLUTTER_SDK}/bin/cache/dart-sdk/bin/dartaotruntime \
   --verbose \
-  --disable-dart-dev ${FLUTTER_SDK}/bin/cache/artifacts/engine/linux-arm64/frontend_server.dart.snapshot \
+  --disable-dart-dev ${FLUTTER_SDK}/bin/cache/dart-sdk/bin/snapshots/frontend_server_aot.dart.snapshot \
   --sdk-root ${FLUTTER_SDK}/bin/cache/artifacts/engine/common/flutter_patched_sdk_product/ \
   --target=flutter \
   --no-print-incremental-dependencies \
@@ -34,18 +34,20 @@ ${FLUTTER_SDK}/bin/cache/dart-sdk/bin/dart \
   -Ddart.vm.product=true \
   --aot \
   --tfa \
+  --target-os linux \
   --packages .dart_tool/package_config.json \
   --output-dill .dart_tool/flutter_build/flutter-embedded-linux/app.dill \
   --depfile .dart_tool/flutter_build/flutter-embedded-linux/kernel_snapshot.d \
+  --verbosity=error \
   package:${APP_PACKAGE_NAME}/main.dart
 
 # Build AOT image.
 ${FLUTTER_SDK}/bin/cache/artifacts/engine/linux-arm64/gen_snapshot \
   --deterministic \
   --snapshot_kind=app-aot-elf \
-  --elf=.dart_tool/flutter_build/flutter-embedded-linux/libapp.so \
+  --elf=.dart_tool/flutter_build/flutter-embedded-linux/app.so \
+  --strip \
   .dart_tool/flutter_build/flutter-embedded-linux/app.dill
-#  --strip
 
-cp .dart_tool/flutter_build/flutter-embedded-linux/libapp.so ${RESULT_DIR}/${BUILD_MODE}/bundle/lib/
+cp .dart_tool/flutter_build/flutter-embedded-linux/app.so ${RESULT_DIR}/${BUILD_MODE}/bundle/lib/
 
```

closed 2024-04-20 by lhoward