sony/flutter-embedded-linux #419: Consider replace systemd with libuv?

Thor-x86, 2024-05-29:
Before I'm working on the PR, I need to know why not use [libuv][1] to poll libinput with event loop? Using systemd on most embedded system is like driving a bus to a picnic site where the only passenger is spouse and the rest seats are left empty. In our case, we included around 25 systemd binaries with some unrelated services including [systemd-machined][2], but we only need sd-event. This is a huge resource waste for embedded system. If you think it's okay to replace systemd with libuv, I'm going to continue finish the PR. Thanks!

[1]: https://docs.libuv.org/en/v1.x/poll.html#c.uv_poll_init
[2]: https://man.archlinux.org/man/systemd-machined.8

HidenoriMatsubayashi, 2024-05-29:
> I need to know why not use [libuv](https://docs.libuv.org/en/v1.x/poll.html#c.uv_poll_init) to poll libinput with event loop?

There is not particular reason and if I remember correctly, weston also uses systemd, which was one of the reasons.

> If you think it's okay to replace systemd with libuv, I'm going to continue finish the PR. Thanks!

Yes, it would be great if your PR could reduce the dependencies on the necessary binaries.

Thor-x86, 2024-05-29:
Alright, I'm working on it. I'll assume libuv is only required if systemd is not exist. Please let me know if you need something

HidenoriMatsubayashi, 2024-05-29:
Thank you.

>  I'll assume libuv is only required if systemd is not exist. Please let me know if you need something

Nothing comes to mind. Currently, libsystemd is only used for DRM backend and poll libinput.

closed 2024-06-02 by HidenoriMatsubayashi

Thor-x86, 2024-06-03:
Friendly reminder to update your Wiki. Unfortunately, we can't pull request (or at least propose a modification) for the Wiki on GitHub.

- https://github.com/sony/flutter-embedded-linux/wiki/Installing-dependent-libraries#only-when-using-drm-backend
- https://github.com/sony/flutter-embedded-linux/wiki/How-to-use-Flutter-for-Embedded-Linux#only-when-using-drm-backend

HidenoriMatsubayashi, 2024-06-03:
Done.