sony/flutter-embedded-linux #424: [Feature Request] Specify DRM connector to use for DRM backend

barribarrier, 2024-09-17:
Hello,

First of all, thank you for the continued development and maintenance of the flutter-embedded-linux project. I’ve been using the Wayland backend and have been very happy with it.

Now, I’m planning to switch to the DRM backend and have a feature request regarding this. Currently, the very first DRM connector is automatically selected https://github.com/sony/flutter-embedded-linux/blob/63b57ca74991ce6782736a1adb8b63928797272f/src/flutter/shell/platform/linux_embedded/window/native_window_drm.cc#L110

In my specific use case, this behavior is not ideal because I need to specify a particular DRM connector, such as HDMI-A-1, rather than the default eDP-1.

Would it be possible to add a feature that allows users to specify which DRM connector should be used for rendering?
Maybe a new environment variable `FLUTTER_DRM_CONNECTOR` just like the current `FLUTTER_DRM_DEVICE`.
My knowledge in C is very limited, so implementing this change myself is challenging.

_Just for reference, how MPV does it:_
https://mpv.io/manual/stable/#video-output-drivers-drm-connector

HidenoriMatsubayashi, 2024-09-18:
> Would it be possible to add a feature that allows users to specify which DRM connector should be used for rendering?
> Maybe a new environment variable FLUTTER_DRM_CONNECTOR just like the current FLUTTER_DRM_DEVICE.

Yes. Thanks for the proposal.

> My knowledge in C is very limited, so implementing this change myself is challenging.

I can review your change :)

barribarrier, 2024-09-19:
`weston --backend=drm` works too.
Looking at `wetson`, `mpv` sources, I can't detect any obvious issue with gbm surface creation.

These errors are weird:
```
==331207== Warning: invalid file descriptor -100 in syscall close()
==331207== Warning: invalid file descriptor -100 in syscall close()
../../flutter/third_party/dart/runtime/vm/virtual_memory_posix.cc: 69: error: mmap failed: 22 (Invalid argument)
```

barribarrier, 2024-09-19:
Found the issue here: https://forums.developer.nvidia.com/t/gbm-surface-create-fails-if-flags-0/279951
Setting `flags` to `0` works. No more segmentation fault.

closed 2024-09-20 by HidenoriMatsubayashi