sony/flutter-embedded-linux #433: Pull request info in README is out of date?

jamesf-carallon, 2025-02-21:
Hello. I can see in the pull requests section that PRs have been accepted from various sources (I'm guessing not just within Sony). But in the README it says

> Now, we cannot accept any Pull Request (PR). Because We are building a system (e.g. CLA) to accept PRs, so please wait for a while the system is getting ready! However, we are always welcome to report bugs and request new features by creating issues.

>With the assumption, our final goal of this software openly is to be merged this embedder into [Flutter Engine](https://github.com/flutter/engine) after getting feedbacks. And [Google CLA](https://cla.developers.google.com/about/google-corporate) will be required when we do that in the future. Therefore, we cannot easily accept an external PR. However, you can free to create issues for reporting bugs and requesting new features.

Please could you provide clarity on whether PRs are welcome?

HidenoriMatsubayashi, 2025-02-24:
@taka-okayama

taka-okayama, 2025-02-28:
@jamesf-carallon @HidenoriMatsubayashi 
Sorry, let me check the PR policy.
However, at this time, we would appreciate it if you could follow the wording in README.md.

jamesf-carallon, 2025-03-11:
@taka-okayama 
Thank you for getting back to me. I have an implementation of software rendering (#287), which I'm tidying up and soon will be suitable for a PR, if you guys would like it. If your policy changes and PRs become welcome, let me know.

zxcv1884, 2025-05-09:
@taka-okayama 
Hi, I saw the PR is still open. Any updates on when it might be merged?

taka-okayama, 2025-05-09:
@zxcv1884 

sorry for inconvenient.
I've just merged! https://github.com/sony/flutter-embedded-linux/pull/434

felixlechat1, 2025-08-25:
> [@taka-okayama](https://github.com/taka-okayama) Thank you for getting back to me. I have an implementation of software rendering ([#287](https://github.com/sony/flutter-embedded-linux/issues/287)), [...]

Hi @taka-okayama, any chance this PR is ready to review or is already active? I am currently trying to make a flutter application render directly to a framebuffer device, haven't found much support for this so far...